rstephan/ArtnetWifi #1: ESP-01 issue

nccwarp9, 2016-05-30:
Im trying to get this to work on ESP-01 and have issues. InitTest works but when trying to send via Jinx or Glediator no joy. 

Any specific setting need to be set up ?

rstephan, 2016-05-30:
Hi,
if you are using the ArtnetWifiNeoPixel example unmodified, then it will not work, that's right. Jinx! can only send 512 bytes to one node.

> Quote:
> "You also can set the amount of channels to be transmitted. Art-Net, as DMX based protocol, supports  maximal 512 channels per node. If you set the Channels value greater than 512 it will be corrected to 512 on saving. The Art-Net specification allows sending less than 512 channels, so you can choose lower values."

The example has 240 pixels, that times 3 (RGB) equals 720 bytes.
Is your matrix as big as that? Try to reduce numLeds in line 17 to something less than 170 pixels.
If you really have that many pixels, you must transmit bytes 0-511 to universe 0, 512 to 1023 to universe 1, and so on ...
I think Jinx! wont work this way, if not correct me please.

nccwarp9, 2016-05-30:
It works when changing the led number to 170. Thanks!
What software can send more than 512 bytes ?
My matrix is 340 pixels.

rstephan, 2016-05-30:
As it turns out, after more time with Jinx!, you can add more output devices, even they are the same type. So add a Art-net for universe 0, then a second one with universe 1 (need more? add these in a row). After that you have to patch the channels to the output (Setup -> Output Patch), ask the documentation how to do that.

closed 2016-05-30 by rstephan