rstephan/ArtnetWifi #2: 5v and 3.3v matching for data pin?

sashabeep, 2016-07-25:
Hello, can you provide some information about connecting ESP with 5v(or even 12v) stripes? Is 3.3v is enough for data and i can just use dc-dc stepdown converter for ESP VIN pin?

rstephan, 2016-07-25:
If you are using a "bare" ESP-01 to ESP-11 module, you have to supply 3.3V to power the controller. In the case of you are using an more "advanced" board with an onboard regulator, even 12V should be fine, for example the NodeMCU board has one. But check the specifications for the regulator In doubt use an external dc/dc converter to generate 3.3V (500mA to 800mA at least).
To connect simple RGB(W) stripes 5V, 12V or what ever, you have to build a driver for each color. The capabilities for the driver depend on the number of LEDs.
Smart Pixel/WS2812/NeoPixel for example are 5V-only LEDs, the data signal should be 5V too, but in most cases 3.3V will also work fine, To protect your ESP from magic smoke, put a series resistor (100 Ohm or so) in line with esp-data-out to led-data-in.

sashabeep, 2016-07-26:
Thanks for advice!

closed 2016-07-26 by sashabeep