rstephan/ArtnetWifi #3: Ws2812

darknessatnoon, 2016-09-15:
Is it possible to use this code with ws2812b?

rstephan, 2016-09-15:
Take a look at the NeoPixel-example
https://github.com/rstephan/ArtnetWifi/blob/master/examples/ArtnetWifiNeoPixel/ArtnetWifiNeoPixel.ino

darknessatnoon, 2016-09-15:
I see already.
When i already flash this to ESP8266. Wifi connection is ok and comes up in my network.
const byte dataPin = 2; means GPIO2 ?
On Arduino Uno i get ws2812b to work with Adafruit_NeoPixel strandtest.
Adafruit_NeoPixel strip = Adafruit_NeoPixel(60, PIN, NEO_GRB + NEO_KHZ800);  Working on Arduino
When i flash this to ESP8266 on GPIO2 nothing happens with (#define PIN 2) =GPIO2.
I try on ESP-01 and ESP-12F with same result.

rstephan, 2016-09-15:
Sorry, i can't see a problem.
"dataPin = 2" is GPIO2, so the example should work out of the box, even on a ESP-01 / -12
The code was designed on the "nodemcu" platform. Only there is GPIO2 == D4.
I have tested it on a ESP-01 with a scope (no real LEDs). All pulses came out right after the wifi-setup.
Have you used 5V for the LEDs? The WS2812 is 5V only. If there is all at 3.3V the CPU will come up, but the LEDs will not light up. So, ESP @ 3.3V and WS2812 @ 5V. For safety, put a 100 ohm resistor in the data line from the CPU to the first LED.

darknessatnoon, 2016-09-16:
Yes i know.
I have connected my esp 12 usb version to pc usb.
The ws2812b is connected to extra 5v power supply. Only ground and d4 to Din is connected between esp and ws.
Yesterday i connect normal led to ground and d4. When i connect to esp from jinx over artnet, the led starts flickering. Ok some Data coming.
Maybe problem with timing.
I see on diffrent project connect a cap 100pf to ground and d4. Also not working. Resistor also didn't work.

rstephan, 2016-09-16:
If D4 starts flickering, so artnet and pin-labeling should be fine.
Have you tried the Adafruit_NeoPixel strandtest on a ESP8266? Try every piece alone. 
You can compile the ESP8266 project with different speed options. Is everything (ESP8266, NeoPixel, ...) up-to-date?
Try again the strandtest with the Arduino Uno, bring it to work, then switch to the ESP and upload the same sketch, if this is not working, there is the problem.

darknessatnoon, 2016-09-17:
Ok now its working.
I dont know why but it dont like different voltages.
The esp works with 3,3v and ws 5v ok.
When i connect the ws to 5v and the DI signal is 3,3v (LD1117) its not working.
When i connect all to 3,3v its working. I have also tried with Level Shifter no chance.
Seems the DI signal must be same voltage than ws +.
Arduino Uno works without problems.

closed 2016-10-22 by rstephan