rstephan/ArtnetWifi #4: ESP8266 Library

Capii1, 2016-10-21:
Where do I find the ESP8266 Library that includes ESP8266WiFi.h file. I've downloaded what I thought was the correct library however the build still fails first as it is missing the file "#include <functional>"

rstephan, 2016-10-21:
The library is included inside the esp8266-board package:
Linux: ~/.arduino15/packages/esp8266/hardware/esp8266/2.2.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h
Windows: C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src\ESP8266WiFi.h

Be sure that you have also selected a ESP-Board. If it's accidentally set to "Uno" it won't work.

Capii1, 2016-10-24:
Do you happen to know how I can get it for Mac? I'm also slightly confused on the board selection. If I'm putting the ArtnetWifi sketch on the Uno with the ESP8266 connected to the uno shouldn't I have Uno selected? It is the main microprocessor which will then control the ESP8266 to connect to my network? Im currently using the ArtNet Master sketches to control my led lights along with an Ethernet shield and that is how it run that sketch. I would assume the ESP simply replaces that shield and uses the ESP8266WiFi.h file instead of the Ethernet one. Sorry if that was long winded, your help is greatly appreciated.

rstephan, 2016-10-25:
There are two ways, to use the ESP board. 
- The first (i call it the old way) is to connect a ESP module to the serial port of an UNO, Leonardo or what ever and use AT-commands to handle the WiFi stuff. This is what you want to do. 
- The second way (new one) is to replace the firmware of the ESP board direct, no need for an extra microcontroller. This is what my library needs.

To replace the firmware of the ESP-board you need a "board-package" for your Arduino IDE. This is because you are now using the CPU on the WiFi module to do stuff for you. 
I use a board called "NodeMCU", it is basicaly a ESP and a USB-to-serial adapter on one board. A ESP-01 and an external serial-adapter can to the same, no need to buy something new.

My library was specificaly desined to run on a bare ESP, so it is requesting the ESP8266WiFi.h header file out of the esp board package. 
If you want still use "the old way", i am sorry, you can't use my library.

(Linux, Windows, Mac: if you have the board-package and my library, it will compile on all platforms.)
(Mac: `/Users/<username>/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/ESP8266WiFi.h`)

Capii1, 2016-10-25:
Thank you so much for your help! I completely understand now. I've ordered a [NodeMCU](https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) board to work with. I've also installed the board manager and verified that the code compiles correctly and it does! I'll also be looking into flashing my ESP-01's to the NodeMCU firmware as I ordered several of them for this project and I'd like to be able to make use of them.

Capii1, 2016-10-27:
@rstephan I was able to flash the ESP-01 and get the sketch loaded on the board. I can see it receiving the artnet data via the RX led on the uno. (I'm using the uno only as a passthrough to flash and power the ESP-01) my final question is do you have any idea what pin I need to use to send the digital signal out? I've tried a few things and haven't had much luck with this last step.

rstephan, 2016-10-27:
What do you mean with "send the digital signal out"?
I am guessing, you are using WS2812 LEDs? Have you seen my closed issues?
https://github.com/rstephan/ArtnetWifi/issues/3
https://github.com/rstephan/ArtnetWifi/issues/2
Something usefull in there? No, go on and give me more details.

CR77777, 2017-01-18:
Hey, I have a question, is it possile to use your library for the Pretzelboard ? This should have the same WIFI Chip on Board. Maybe you can help me to match ? Thanks a lot

rstephan, 2017-01-19:
The Pretzelboard is the "old way", see [early post](https://github.com/rstephan/ArtnetWifi/issues/4#issuecomment-256060330). You can chop off the ATmega and use the ESP alone. But don't use an hacksaw! Desolder the ATmega, connect the rx and tx from the CH340 USB-serial converter to the ESP. Now you have an expensive variation of a NodeMCU board.
I suggest, put the Pretzelboard aside, and buy a NodeMCU (about 10 Euro from china).
Rewriting the lib to use the two controller, and utilise the poor serial connection between the two, is not worth. If you need something to learn coding, do it.

Nikoskir, 2017-02-19:
Hello,
first of all thanks for this info about artnet and esp.
It helped me a lot and i've managed to make a network with 10 esp8266 attaching 30 ws2812 on each one.
Using Qlc+ and a good access point i made really nice effects. Some of them working on different universes
and the result is fabulous.
I have some resets from time to time but its ok .
After some reading about esp resets i can tell its not for now so urgent so i dont mind.
Too many opinions but none of them accurate.
The setup i have now requires a PC.
My question is if its possible to use artnet transmit with a standalone setup as : 
1) arduino uno or mega running neopixel patterns
2) ethernet shield with arduino
3) an access point capable of many clients 
4) esp8266  X 10 having the neopixel strings

thanks in advance

Niko

rstephan, 2017-02-20:
The short answer is: no. My library was only designed for the ESP8266, so Uno, Mega and so on are out. But you can use another ESP8266 to generate the patterns. My library is able to transmit ArtNet as well.
If you still want to use ethernet you have to write your own ArtNet transmiter. One UDP-frame with a little header, that's it.

Nikoskir, 2017-02-20:
Hello 
actually is not that i want to use ethernet exclusively.
my thinking is if an esp8266 server can handle 10 clients.
But as second thought i can make an esp server connect to an access point 
which will work as repeater and the clients connect to the access point.
i am just a hobby man who likes to read and put things down.
i will try like this and see if its working.

thanks again

Niko

rstephan, 2017-02-20:
I don't know the upper limit of the ESP8266, but if you need power you have to choose a fast "Arduino". A Uno runs at 16 MHz, a ESP8266 can do 80 MHz. The final performance depend on many reasons: 
- generating patterns is CPU-time,
- move the data from the CPU to the network-controller(LAN/WiFi) takes time as well,
- transmit from A to B and
- finaly the receiver has to handle the data.

10 clients and 1 "frame/s", sure a Uno can do this.
10 clients and 50 "frames/s", maybe the Uno can't generate so many image data (this is only number one on my list).

Build it up and make some tests. That's all i can say, sorry.

TJFOE, 2021-07-14:
I have a basic question: I want to control my DUE and ESP8266 module in communication with my own Windows app on the tablet via WiFi (only send and receive volume values for a DigiPot/Rotary encoder in an amplifier). Is ArtnetWifi suitable for this? I am unclear at the moment in what way I make the library talk to Windows...? I would be very grateful for a hint.

Regards
Thomas