rstephan/ArtnetWifi #5: UDP not broadcasting

tonyschuite, 2016-11-21:
Hi there, 

I am trying to get this to work on a NodeMCU Board.
so far so good, it's just that I can't get it to broadcast the Artnet Node.

in QLC+ I can manually set the IP, but I cannot do this with other software.

any suggestions?

rstephan, 2016-11-21:
I need a little bit more information. What do you mean with "not broadcasting"?
The lib can be receiver or transmitter but is not a node. 
If i have to guess, you are using the receiver to light up some stuff. In this case you have to tell your app/tool where to send the art-net data. So you have to enter the IP of all your node(s) into it.
This ArtnetWifi is a simple library.
But take a look into my second project https://github.com/rstephan/ArtnetnodeWifi this is a node and it can response to art-net poll requests so your app/tool can "autodetect" all nodes inside your network.
The ArtnetnodeWifi is also simple but it should be sufficient to do the job.

tonyschuite, 2016-11-22:
the Node branch was actually the solution. This was exactly what I was talking about.
[clodes?]

closed 2016-11-22 by tonyschuite