rstephan/ArtnetWifi #6: Max WS2811

clemvixen, 2017-02-27:
What is the number maximum of WS2811 in ESP8266 with artnet ?

thanks :)

rstephan, 2017-02-28:
First of all, there is no simple answer to this question. The next thing is, this is a Art-Net library. I don't know what you are connecting to it. A universe can have 512 channels. The lib can handle multiple universes. The upper limit depends on frame-rate and memory (RAM).

The ArtnetWifiNeoPixel example is what it is, a example to show you possibilities, not a high performance implementation. You have to call Adafruit for the performance of the NeoPixel-lib (WS8211, ...) [Adafruit_NeoPixel](https://github.com/adafruit/Adafruit_NeoPixel)

But as a small hint, from feedback, i know a configuration with 2 universes and about 340 pixels. Keep in mind, 340 pixels draw about 20A of current, that is a big power supply!

closed 2017-02-28 by rstephan

clemvixen, 2017-02-28:
Thank you for your reply !! I understand better !!

Have you a method to calculate the number max, for example if I have a
microcontroler with this feature : RAM = 64kbyte and frequency : 120mhz.
And I would like 30fps

It's possible to calcul this number maximum ?

Thank you very much :)
​