rstephan/ArtnetWifi #10: Wifi artnet connection

pinq-, 2017-11-03:
Hi,

I tried yesterday your program "ArtnetWifiNeoPixel.ino". It connected to router and it also did do the initTest and I can see it in my router so connection is there and led controlling works. But when I try to send data to the wemos D1 mini lite, nothing happens. I tried with jinx and q light control plus. Ip is correct and universe is correct. In stead the routers wifi connection crashes and then disappears. 
Any good ideas what happens or how to debug this? 
Otherwise great code! Clean and understandable.

rstephan, 2017-11-09:
Just a guess, but most likely a power supply issue. Is the WiFi connection stable in idle? Try with just one LED. 
Or try the ArtnetWifiDebug example, user the terminal to check your network and program setup. If you can see solid data and only the NeoPixel don't work it's a power supply problem. Even 10 LEDs can add up to 600mA extra, keep this in mind.

rstephan, 2018-10-27:
@MFClark 
Try the `ArtnetWifiDebug.ino` example.  Check the WiFi connection and the serial monitor output.

Example output:
```
....
Connected to xxx
IP address: 192.168.x.y
```
This is also the output of `ArtnetWifiNeoPixel.ino` too! It will show up before `initTest()`. If not, check the USB connection and port-speed.

rstephan, 2018-10-27:
Maybe a problem with the HUZZAH. I have tested with my Feather HUZZAH and i can't see any problem.
I don't know your setup, but the HUZZAH has three power inputs (USB, battery and a pin). If you just connect the battery it is hard to read the serial ouput. My board need 5 seconds and i can `ping` it.
If your power is the USB connection, the serial monitor will not "autoconnect" the the device. You have to close and open the serial monitor, and be sure to grab the right COM-port.

MFClark, 2018-10-28:
Thank you, that was it!

closed 2018-10-29 by rstephan