rstephan/ArtnetWifi #11: ws2811 led fliker

TimoteoMendes, 2017-11-17:
Hello,
I have been testing your code using the resolume or touchdesigner.
Before the connection is established is all ok just after the connection the leds start to flick and changing the position as well as colors.
Do you have a clue what can it be?
Thanks

rstephan, 2017-11-19:
First guess, it is most likely a power supply issue. One LED = 60mA, 10 = 600mA, 100 = 6000mA ! and so on. Try the  [Strandtest from Adafruit](https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/strandtest/strandtest.ino) it is available from the Arduino IDE as well. If this isn't working properly, than my code has no chance.
Can you send a single "frame"? Is the ESP allways and stable connected to the WiFi network?
How many LEDs are connected? I hope this helps.

TimoteoMendes, 2017-11-19:
Hello Stephan,
The power supply is a good and powerful enough for this task.
Probably i did not express my self correctly.
The test was done in the begining to identify the colors because it was not correct and that correction was done in the Resolume.
What happen is when nothing is send, the strip turns on a led in a strange but sequential loop also changing color.
The strip used is a ws8211 (3 leds for a single chip).
A single frame is possible no problem on that.
The network is done using a dedicated modem for this and i seems stable to me.
There are 50 pixels or 150 leds.
Thanks

rstephan, 2017-11-19:
Strandtest is working, right?
For me LED equals pixel, so 50 pixels-> 3A current, okay.
The ESP boots up, you are sending a (or more) frame(s). Than you stop, and after that the pixels start to do crazy things. The ESP8266 runs the `ArtnetWifiNeoPixel` program?
Under normal conditions i would say there is a nother server on the network that is sending out artnet packets.
Can you test the `ArtnetWifiDebug` program, to see if nothing else is transmitting?
The pixels should maintain their last color, if not something else is driving the data pin. Have you checked the connections? How many pixels are crazy? Just one or one half of the strip? I need more data...

TimoteoMendes, 2017-11-22:
Hello,
The program is that one and it work ok besides the first pixel is always blue...
The problem is really when the data is stoped the led strip start the strange behaviour. I have a video of that but don't know how to post here.
Can I send it to an email?
Thanks

rstephan, 2017-11-22:
> the first pixel is always blue...

In  Strandtest too? 
I do only data translation, the pixels are driven by the adafruit library. Are the pixels broken? Bad connections? Can you use other LEDs? Can you disconnect the data-pin after the transmission is stopped? As i said, the pixels should maintain their color. If they do the strange things again there is may be a bad connection.
To see my email do a `git log`.

TimoteoMendes, 2017-11-22:
In strandtest that does not happen.
After the strip and the esp is turned on the sequence of tree colors is done all Leds turns of and wait a signal.
Then when the artnet Channel is open the first pixel turns blue and start the strange behaviour.
Have already tested different strips but that behaviour persists.
If the artnet communication Channel is closet the strip keeps the first led on and the one that das blinking
Can you explain the git log?
Thanks

rstephan, 2017-11-22:
What does _channel open_ mean, have you tested `ArtnetWifiDebug`?
After you have _closed the channel_, are all LEDs solid?

github was an idea after `git` the source code management tool. You need the git-tool to see the `git log`. Only a zip-file is not enough.
Get a copy of [git](https://git-scm.com/), it is available for all platforms, there are portable versions as well.
After that: 
```
git clone https://github.com/rstephan/ArtnetWifi.git
cd ArtnetWifi
git log
```

jamesheaver, 2017-12-19:
I'm having what appears to be the same issue when running on an esp32. Was there any resolution to this?

rstephan, 2017-12-21:
I am waiting for a response as well.
Have you gone throu all my suggesstions?
For me, it is a pixel and/or power-supply problem.
I am not responsable for the NeoPixel-Library. 
Try the `ArtnetWifiDebug` example.
Sorry, there is not much i can do for you.

jamesheaver, 2017-12-21:
I've tried strandtest which works fine.
I've tried artnetwifidebug and that works fine
output looks like this:
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0

Connecting to WiFi
Connecting...
Connected to XXXX
IP address: 10.139.164.106
DMX: Univ: 0, Seq: 1, Data (3): 0 0 A0 
`

I've even tried combining the output and debug code and I can see the artnet data coming through, but the leds jsut flash randomly with the first one lit. It doesn't seem to matter what data is being sent, even if debug is outputting '0 0 0', then the leds still flash randomly (other than the first one which stays on all the time).

rstephan, 2017-12-22:
Maybe we have a memory overflow problem.
How many LEDs have you connected to the ESP.
What program are you using? What are your settings?
You have the ESP32, right?
TimoteoMendes didn't say what he is using.
What Arduino version?
What Adafruit NeoPixel version? (In Arduindo Sketch->Library Manager (or something like this ...))

jamesheaver, 2017-12-22:
I have tried it down to just 2 leds.
I am running through Jinx. This is the first time I've played with DMX, so if there is a known good way to test this, let me know.
[image]

Yup, ESP32. I've tried two different dev boards, both with the same result.

Arduino 1.8.5
Adafruit neopixel 1.1.3

rstephan, 2017-12-22:
The NeoPixel library has no proper support for the ESP32. There is a open issue https://github.com/adafruit/Adafruit_NeoPixel/issues/139
There are timing/interrupt issues if WiFi is enables. This is why the strandtest is fine (no WiFi). 
Sorry. 
(I will remove the ESP32 support from my example until the problem is fixed.)

jamesheaver, 2017-12-23:
ah, good to know that I wasn't just doing something silly.

thanks for running it down for me.

RBasile, 2022-06-24:
Hey, 
I did some tests with the NeoPixel example and a esp32 and had no flicker
the support for the ESP32 and Adafruit_NeoPixel is probably fixed since the change made here https://github.com/adafruit/Adafruit_NeoPixel/commit/cd0b2217b2dfa298c39049dad8d512cf87d82ab0 that uses the RMT peripheral?

rstephan, 2022-09-28:
@RBasile thanks for the info. Sorry for the delay :-(

I have added back support for the NeoPixels in the example.
Thanks Adafruit and their supporters!

closed 2022-09-28 by rstephan

dzelionis, 2023-04-07:
go for https://github.com/Aircoookie/WLED it supports all the protocols you need ... very powerful...