rstephan/ArtnetWifi #15: Network timeout

blackrockcitynv, 2018-07-16:
Code seems stable but when the node connects without artnet data being sent it renews its lifetime counter for 20:00 if i send artnet data the timer never renews and eventually the unit disconnects and reboots. If I ping the unit then the timer stays live and connection stays stable. Do you know why it does this?

rstephan, 2018-07-26:
Sorry, was a bit busy.
I don't know the issue. Maybe a problem with you WiFi access-point? A energy savings option?
What is 'lifetime counter' for you? Is it in your router or the ArtNet software?

rstephan, 2018-07-27:
With this setup, energy savings settings are more likely than before. But, it is just a guess.

smcd253, 2020-08-14:
I have also been experiencing this problem. I have multiple nodes running this code and they all reset at around 20 min. Is there a way to disable this counter and/or set it to infinity?

vepkis, 2023-01-09:
I solved this in the following way:

`// connect to wifi – returns true if successful or false if not
bool ConnectWifi(void)
{
  bool state = true;

  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
  return state;
}`

The modification is not very clean. Surely this function can be improved.