rstephan/ArtnetWifi #16: ws2812b flickering every 1sec +-

vettorazi, 2018-08-27:
hey guys!
first of all:
thanks for the library!
it's the first that i tried that works with esp8266 and ws2812b. (others the stip flickers alot)
this one works great but sometimes (every 2 or 1 second +-) the strip blinks.
anyone has the same issue?

rstephan, 2018-09-05:
It is allways hard to tell from a distance.
But, how many LEDs are connected? 1 or 2 LEDs or maybe 100. More LEDs need a bigger power suppy.
Try the `Standtest` from Adafruit to check wireing and your power suppy.
Something wrong with your ArtNet transmitter?

cloverstreet, 2020-12-26:
I am currently having same issue.  I've checked with multiple art-net transmitters.  I've removed everything except for simple short print to screen commands.  About every 1 seconds, the art-net data streams stops.  Like the esp32 is taking a quick little break.

rstephan, 2020-12-30:
@cloverstreet 
Flickering LEDs is different from your problem. You don't get any data once every 1 second. Because of this I suspect a network problem. Bad WiFi? Bad ESP-Antenna? 
Have you tried my `ArtnetWifiDebug` example? Is the frame-sequence in order or are there any missing frames?