rstephan/ArtnetWifi #21: Not working with ESP32 ethernet

DatanoiseTV, 2019-03-27:
I am using the Olimex ESP32-EVB board and I am having trouble making the library work.

My code:
```
#include <WiFiUdp.h>
#include <ArtnetWifi.h>
#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      Serial.print("DNS:\t\t");
      Serial.println(ETH.dnsIP());
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

#include <NeoPixelBrightnessBus.h> // instead of NeoPixelBus.h
// LED Strip
const int numLeds = 288; // Change if your setup has more or less LED's
const int numberOfChannels = numLeds * 4; // Total number of DMX channels you want to receive (1 led = 3 channels)
#define DATA_PIN 2 //The data pin that the WS2812 strips are connected to.
NeoPixelBrightnessBus<NeoRgbwFeature, Neo800KbpsMethod> strip(numLeds, DATA_PIN);

// Artnet settings
ArtnetWifi artnet;
const int startUniverse = 0;

bool sendFrame = 1;
int previousDataLength = 0;

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  Serial.println("Received DMX frame on universe " + String(universe) + String(" with length ") + String(length));
  sendFrame = 1;
  // set brightness of the whole strip
  if (universe == 15)
  {
    strip.SetBrightness(data[0]);
  }
  // read universe and put into the right part of the display buffer
  for (int i = 0; i < length / 4; i++)
  {
    int led = i + (universe - startUniverse) * (previousDataLength / 4);
    if (led < numLeds)
    {
      strip.SetPixelColor(led, RgbwColor(data[i * 4], data[i * 4 + 1], data[i * 4 + 2],  data[i * 4 + 3]));
    }
  }
  previousDataLength = length;
  strip.Show();
}

void setup()
{
  Serial.begin(115200);

  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  ETH.config(IPAddress(10, 0, 0, 2), IPAddress(10, 0, 0, 1), IPAddress(255, 255, 255, 0));

  //WiFi.setSleep(false);
  //ConnectWifi();
  artnet.begin();
  strip.Begin();
  strip.Show();

  // onDmxFrame will execute every time a packet is received by the ESP32
  artnet.setArtDmxCallback(onDmxFrame);
}

void loop()
{
  // we call the read function inside the loop
  artnet.read();
}
```

MihirKhara, 2022-03-08:
Hii, I am trying something similar. I'm trying to use ArtNet using ethernet instead of WiFi, were you able to get it working?
If yes, please share the code / how to.
Thanks

rstephan, 2022-03-08:
Hi @MihirKhara 
the example above should be a good starting point.
Can you share your exact problem? Compile-error? Hardware-setup? A code snippet?
I have used the Olimex ESP32-Gateway and it's working just fine.

MihirKhara, 2022-03-10:
Hi @rstephan, I'm using the native ethernet controller on the ESP32. I was just wondering what changes should be done in the example code for it to work with ethernet? Since you have mentioned this in the readme - Note: this library assumes you are using a wifi module. If you can guide me in the right direction, that would be very helpful.
The ArtnetWifiFastLED example is working flawlessly with WiFi, Thanks for the amazing library.

rstephan, 2022-03-10:
Hi @MihirKhara 
Yes, this library is mainly focused on WiFi.
Yes, this library also works with LAN/Ethernet, as I mentioned above. The ESP32-Gateway is also a wired Ethernet device. 
And yes, you have to do a bit work yourself.

* Open the `ArtnetWifiFastLED` example.
* Take a look at the first post (as I mentioned above)
* Copy `#include <ETH.h>`
* Copy all of
```
void WiFiEvent(WiFiEvent_t event)
{
 ....
}
```
* Modify `setup()` like so:
```
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  //ConnectWifi();
```

Please, do **not** reply on this issue! Open a new one, so we can discuss your problem more "privately".
Thank you.