rstephan/ArtnetWifi #23: Changing the type of artDmxCallback

em-esz, 2019-11-03:
Hi,
Have you considered changing the type of artDmxCallback from raw function pointer to 
`std::function <void (uint16_t, uint16_t, uint8_t, uint8_t*)>`?

This would allow more flexibility for the callbacks (e.g. member functions, lambdas). Should be backward compatible as function pointers are automatically converted to std::function. 

I did a quick test and seems to work properly.

rstephan, 2019-11-11:
Why not. I will consider it for my next update.
As long as I can maintain the compatibility with previous versions of the code.

rstephan, 2019-12-09:
Something like this? Take a look at my `ArtnetWifiDebug2` example.
You can use either `setArtDmxCallback()` or `setArtDmxFunc()`.

closed 2022-10-04 by rstephan