rstephan/ArtnetWifi #26: trying to make use of readartnet project

derrgis, 2020-05-07:
Hello,
I had great success using your ArtnetWifi lib on ESP8266.
I'd like now to use this https://github.com/derrgis/readartnet with ESP32.
Based on OpenFrameworks it reads a video, automatically resize the matrix and convert to artnet.
(Video columns are universes)
So I wonder how it would be possible to use multiple pins on a ESP32 to send each universe (video column as said) to different ledstrips.
Is that possible with your ArtnetWifi ?
Thank you.

derrgis, 2020-05-08:
This is not an answer to my issue but unfortunately even simple tests with ESP01 + WS2812 give flash problems. As described here :
/*https://github.com/rstephan/ArtnetWifi/issues/20
**https://github.com/rstephan/ArtnetWifi/issues/11
**https://github.com/rstephan/ArtnetWifi/issues/16
*/
So I stop now with this.

rstephan, 2020-05-10:
Just a side-note. Flashing or flickering is a setup-problem:

* bad power-supply
* low quality LED strips
* poor WiFi network
* Adafruit or FastLED library perform differently on ESP8266 or ESP32

Is is not a limitation of the ArtnetWifi library. Outputting different universes on different pins is not a problem, but don't expect high frame-rates.