rstephan/ArtnetWifi #29: Miss synchronizing

Nikitok122, 2020-06-19:
hello!!! having this problem in jinx software 
https://www.youtube.com/watch?v=pDtPCQQSfgs

rstephan, 2020-06-19:
I need more info! Jinx doesn't work is not a good start. Is it a problem with my library or is it a fault from your software?
* What Arduino hardware are you using?
* What is the green device in the video? It's not an Arduino!
* What code is running? Some of my examples?
* How "good" is your WiFi?

**But one guess:**
16x16 RGB LEDs are 768 Byte in one stripe?
One Universe with 170 LEDs per Art-Net UDP-Frame?
Update-Order:

1. Universe 1 -> Update -> Universe 2 -> Update ...
2. or. Universe 1 -> Universe 2 -> Update ??

In case 1. maybe one frame is dropped, if the Arduino is busy writing to the LED strip.
Try buffering both frames and make one update after.

Nikitok122, 2020-06-19:
Thanks for response!! I'm using esp32, that green device just a wifi router, which is between PC and esp32. Yes, that martix display contain 2 universes. Yesterday I tried gladiator software and seems no problem like this, but there is another problem with synchronization but in the whole screen, looks like dropped frames. Tonight I'll check sending packets and will see how packets serv universs: one packet to one universe or one packet to all universes

Nikitok122, 2020-06-21:
I found the reason of that behavior, there were my router. so weird. but framedrops still present in any softwere