rstephan/ArtnetWifi #32: More than 512 leds, no light comes through

davidboon78, 2020-11-13:
I'm using an ESP32  and trying to get 1024 LEDs to run, but if I make numLeds more than 512, even 513, the entire light array goes black.  Some investigation finds that onDmxFrame sets sendFrame to 0 every time it's called.  I tried to eliminate sendFrame but the FPS seems to drop significantly.  Any suggestions?

const int lastUniverse=startUniverse+maxUniverses;

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  if(universe<startUniverse||universe>lastUniverse)
    return;

  // read universe and put into the right part of the display buffer
  for (int i = 0; i < length / 3; i++)
  {
    int led = i + (universe - startUniverse) * (previousDataLength / 3);
    if (led < numLeds)
      leds[led] = CRGB(data[i * 3], data[i * 3 + 1], data[i * 3 + 2]);

  }
  previousDataLength = length;
  FastLED.show();
}

rstephan, 2020-11-14:
Here are my questions/advices for you. (I am guessing, the `ArtnetWifiFastLED`-Example is used.)
* Is the `initTest()` working? Show all LEDs the red/green/blue pattern at start?
* Removing `sendFrame`-behavior is a bad idea! With 512 LEDs your frame rate will drop to 1/3, with 1024 to 1/6!
* The `FastLED.show();`-call takes time. You can find some calculations here: https://github.com/rstephan/ArtnetWifi/issues/31#issuecomment-674367962
* Send the universes in a row, with no gap, e.g.: 0,1,2 `wait` 0,1,2 `wait`. **NOT**: 1,3,4 `wait` 1,3,4

I will try to replicate your setup, but this will take some time.

davidboon78, 2020-11-15:
- the test works fine, all leds light up.
- yeah that's what I'm finding.
- I'm using resolume to send the universes.  Each one with 510 channels.  

Thanks for any help.  If I can solve it myself ill let you know.