rstephan/ArtnetWifi #33: Is it possible to manually set the IP Address?

star-dust-1, 2021-04-11:
Just curious if there's a variable I can use to set the IP Address of my ESP8266 board manually in the code instead of the board choosing an address automatically? I'm having the issue where if I reset my modem and the board the IP Address changes and then my other software that's trying to point to it can't find it anymore. 

Thanks!

TrigpointingUK, 2021-04-11:
There's probably an option in your router's DHCP config which will ensure that your ESP8266 is always given the same IP address, based on its MAC address.  Or you can set a static IP address in your code (best also set gateway, subnet etc while you're there...)
```
IPAddress local_IP(192, 168, 1, 184);

IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(8, 8, 8, 8);
IPAddress secondaryDNS(8, 8, 4, 4);
```

star-dust-1, 2021-04-11:
I'll give it a try, thanks!

closed 2021-04-17 by rstephan