rstephan/ArtnetWifi #36: startUniversity > 2 causes problems

ripatee, 2021-09-21:
Hi and thanks for the library!

I'm having some problems with my current setup. I have a setup containing 6 ESP32s and each of them have 240 pixels of WS2812. My plan was that 1st ESP32 uses 0 and 1st universum, 2nd ESP32 2nd and 3rd and so on, as 240 pixels are out of limits of 1 universum. My problem is, that if I set startUniverse higher than 2 I'm unable to get ESPs working. If I change startUniverse back to 0, it starts working. Now two of my units are working (startUniverse 0 and 2). I'm using MadMapper to send the ArtNet.

I also tried to set startUniverse back to 0 and send ArtNet packets as unicast, but that didn't work neither. But it worked instantly when changed back to multicast (EPSs were receving the same data of course). Is there some special requirements when using unicast ArtNet with ArtNetWifi?

I also noticed that ESPs were unable to connect to wifi when the ArtNet stream was on. There was also a huge latency when pinging ESPs during the ArtNet stream, over 3 sec. Is this normal behaviour?

 I'm using some regular ESP32 dev boards from Ali Express and the code is basicly the ArtnetWifiFastLED example with static IP.

Thank you for your time!

rstephan, 2021-09-24:
Hi,
you are right. The calculations in the example (Line 100 and 117) are wrong. 
I will fix this.
Thanks for the hint.

rstephan, 2021-09-24:
I have made a first fix.
Can you try it please. I have no LED setup to test with.