rstephan/ArtnetWifi #38: LED strip stops working after some time with high FPS

tpimh, 2021-10-08:
I'm currently using ArtnetWifiNeoPixel example on Arduino Nano 33 IoT with a strip of 46 LEDs.

After a successful connection to Wi-Fi, the animation of the LEDs is smooth and correct for some time, but after ~20 seconds it starts lagging, then all LEDs turn off. If the board is rebooted, it connects to Wi-Fi again, and works well for another ~20 seconds.

I discovered that with lower FPS values, the time before the lag begins is increased. With 9 FPS the time from power on to the state where no LEDs turn on anymore is ~2 minutes. With FPS values of 6 and lower this doesn't happen at all.

What can possibly be causing this? I tried several Wi-Fi routers, and it doesn't seem to be the problem.

rstephan, 2021-10-08:
From the description, it sounds like an out of memory/overflow problem.
The Nano 33 IoT uses a two controller architecture, so code is executed on different part of the board.
Or maybe it is a simple capacity problem in terms of bandwidth. The two controller (ubox & ATSAMD) communicate via SPI.

What is the firmware version of the ubox controller?
Do you use my `ArtnetWifiNeoPixel` example without modifications (except number of LEDs)?
Is there only one Artnet universe with all the LED data?

tpimh, 2021-10-14:
Thank you very much! Testing now, and it's working perfectly!

closed 2021-10-14 by tpimh