rstephan/ArtnetWifi #42: Lag after 30 sec

salutjesuispaul, 2022-04-07:
Hello,

I'm using an ESP32 to control a LED COB with a PWM.
After the upload, everything is fine. But after 30 sec, the data received by the ESP32 in wifi start to lag. I tried with the debugging program but nothing works: after a few seconds, the data received on the serial are jerky. 

I think it's a memory problem, do you have any solutions?

rstephan, 2022-04-07:
It's hard to tell.
How many Art-Net frames per second?
How many Art-Net channels? 
Only one LED, just brightness up and down?

Can you share your code, maybe a snipped?
The `onDmxFrame()` should only be a one-liner, am I right?

In many cases it's just bad WiFi.

salutjesuispaul, 2022-04-08:
- I use QLC+ to send Art-net Data ; I think it sends data at 30 fps.
- Only 1 Art-Net channel, in one universe
- Yes, I send fade up and fade down signal.

Here's my code :
`
#include <ArtnetWifi.h>
#include <Arduino.h>

// PWM
const int freq = 5000; // 5000 Hz
const int ledChannel = 0;
const int resolution = 8; // Résolution de 8 bits

// LED COB RGBW
const int ledPinR = 15;
int ledValueR;

// WIFI
const char* ssid = "SFR_C6F0";
const char* password = "ymm32ynxj8cfs23agjr4";
WiFiUDP UdpSend;

// connect to wifi – returns true if successful or false if not
bool ConnectWifi(void)
{
  bool state = true;
  int i = 0;

  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");
  
  // Wait for connection
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 20){
      state = false;
      break;
    }
    i++;
  }
  if (state) {
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(IPAddress(WiFi.localIP()));
  } else {
    Serial.println("");
    Serial.println("Connection failed.");
  }
  
  return state;
}

// ARTNET
ArtnetWifi artnet;
const int startUniverse = 0; // CHANGE FOR YOUR SETUP most software this is 1, some software send out artnet first universe as 0.
const int numberOfChannels = 3; // Total number of channels you want to receive
// Check if we got all universes
const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512) ? 1 : 0);
bool universesReceived[maxUniverses];
bool sendFrame = 1;

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data) {
  
  sendFrame = 1;
  
    // range check
  if (universe < startUniverse)
  {
    return;
  }
  uint8_t index = universe - startUniverse;
  if (index >= maxUniverses)
  {
    return;
  }

  // Store which universe has got in
  universesReceived[index] = true;

  for (int i = 0 ; i < maxUniverses ; i++)
  {
    if (!universesReceived[i])
    {
      sendFrame = 0;
      break;
    }
  }
  if (sendFrame) {
  // PWM
  ledcWrite(ledChannel, data[1]);
  // Reset universeReceived to 0
  memset(universesReceived, 0, maxUniverses);
  }
}

void setup() {
  
  Serial.begin(115200);
  ConnectWifi();

  // ARTNET
  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);
  artnet.begin();

  // PWM
  // Configure le channel 0
  ledcSetup(ledChannel, freq, resolution);
  // Attache le channel 0 sur les 3 pins
  ledcAttachPin(ledPinR, ledChannel);
}

void loop() {
  // ARTNET
  // we call the read function inside the loop
  artnet.read();
}
`

If you think it might be the bad wifi, I would try to create a wifi access point with another ESP32.

salutjesuispaul, 2022-04-08:
I tried to connect my PC & the esp32 to an esp32 wifi access point ; this doesn't work either.

rstephan, 2022-04-11:
There is nothing wrong with the code.
I can reproduce the "error" only if my PC (Art-Net transmitter) is also connected via WiFi.
With a cable connection to my PC I don't see a problem at all. A test ran over many hours flawlessly.

So, as I said, bad WiFi is the problem in my opinion.

HankLloydRight, 2023-07-24:
I'm also having a similar delay problem -- my Artnet transmitter (PC program) is on ethernet, but sometimes it takes 20 seconds or more for the ESP32 to receive the DMX changes.   And then sometimes it's immediate, so it's a very strange intermittent problem.

I'm sure it's not this library, but my question is-- where are those Art-Net packets going for 20 or 30 seconds before the ESP32 receives and processes them?