rstephan/ArtnetWifi #45: DMX-Workshop won't detect this

MartinNohr, 2022-09-06:
For some reason DMX-Workshop and some other apps don't detect this node software. They don't allow manually entering an IP address so it isn't possible to use this software with them. I found one app called ArtNet Controller that does allow manual IP entry. Using that app this node software does work with WS2812B that I tried. The IP assigned from my router is 10.0.0.72 which I entered into the app. Is there some discovery mechanism that needs to be implemented, or how do programs like DMX-Workshop locate DMX nodes on the network?

rstephan, 2022-09-06:
Yes, that's right. Projects with this library are not detectable.
Take a look at my other Library https://github.com/rstephan/ArtnetnodeWifi I guess this is what you are looking for.

MartinNohr, 2022-09-09:
Your other library works perfectly! Thanks so much for doing that work. I used it in this project: https://github.com/MartinNohr/MagicImageWand

rstephan, 2022-09-10:
Happy to help. Nice project as well. Thanks for sharing it.

closed 2022-09-10 by rstephan

MartinNohr, 2022-10-11:
Hey, thanks for the quick response, I’ll look into that one. I really appreciate the effort you have put into this.

Martin