rstephan/ArtnetWifi #47: I found an "ERROR" in your library with the software I'm using.

GabrielDSan, 2022-10-03:
I found an "ERROR" in your library with the software I'm using, could I change the channels that are sent? change the start channel and end channel of the universe?

rstephan, 2022-10-04:
> I found an "ERROR" in your library with the software I'm using

**What is the error?**

>  could I change the channels that are sent? 

**I don't know, I do not know your software.**

> change the start channel and end channel of the universe?

**I don't know, I do not know your software.**

My library is just a "tool" to make Art-Net projects. The examples are only to help and support you during **your** development. In some cases they are good enough for small and/or simple projects, but not for all.

* What is your transmitter software? QLC+? Jinx?
* What is the "error"?

GabrielDSan, 2022-10-04:
> * Qual é o "erro"?
The two software that I tested and they worked, receive 0, 510 channels, but Lumikit SHOW, receives 1.512, and would have to configure to take advantage of the rest of the division of 512 / 3, the 0.666, to connect channels 511 and 512