rstephan/ArtnetWifi #48: receive and transmit simultaneously

samcooler, 2022-12-01:
Hi, I would like to have my ESP32 receive LED data from and transmit sensor data to a base node. Can artnetwifi do this? If so, how? I tried using a single ArtnetWifi object and combining the two examples, and I tried making two objects, one for sending and one for receiving. Neither worked clearly. The base node is a computer using python running stupidArtnet, and I have tested that it can both send to and receive from the ESP32. Just not both at the same time. Thanks!

rstephan, 2022-12-02:
Send and receive should be possible from within one application.
Only use one object! If not, both will fight for the network resource ;-)

There is only one internal buffer for rx and tx data, keep this in mind.

Create one object and call the `artnet.begin("base-node-ip")` method with the destination IP/hostname for your base node in `setup()`.
In `loop()` call `artnet.read()` as normal.
To transmit data, you have to `artnet.setlength(...)` and `artnet.setByte(...)` every time you do a `artnet.write()`.
For the receive part, use the rest from my NeoPixel example.

That should be it. If you have a more specific problem, let me know.

samcooler, 2022-12-06:
That works great, thank you!

closed 2022-12-06 by samcooler