rstephan/ArtnetWifi #49: Const variables ti Int Variables

Juanmi18, 2022-12-19:
Hello, first of all thank you very much for the library, it has served me perfectly so far.

I want to improve my project and add WifiManager (Tzapu). In addition to requesting the Wifi to which the arduino must connect, I want to request the startUniverse and the numLeds through custom fields.

It is giving me the problem that the variables:

numLeds
numberOfChannels
startUniverse
maxUniverses

They are constants and I can't change them to int because universesReceived[maxUniverses]; it gives me the following error:

`array bound is not an integer constant before ']' token`

Can you think of how I could modify the library to be able to set those values ​​as variables and not as constants?

Thanks in advance

rstephan, 2022-12-22:
Hi,

first of all, the `universesReceived` variable is part oy my example, not the library itself.
What you are doing with this example is on your own. I use static allocation.

But you can rewrite it to use dynamic memory allocation:

```
bool universesReceived[maxUniverses];
```
Change the type and add initialization into `setup()` to something like this:

```
bool* universesReceived;
...
void setup() {
...
  // place before artnet.begin(); !!!!
  universesReceived = new bool[maxUniverses];
...
  artnet.begin();
}
```

NOT TESTED!! But should be fine.

Juanmi18, 2022-12-23:
Thanks!!!! Work perfect!!!

I have needed update the length of the led strip with the var of the json with this:

leds.updateLength(numLeds);

This is the final code:

```
/*
This example will receive multiple universes via Artnet and control a strip of ws2811 leds via 
Adafruit's NeoPixel library: https://github.com/adafruit/Adafruit_NeoPixel
This example may be copied under the terms of the MIT license, see the LICENSE file for details
*/

#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#if defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#else
#include <ESP8266WiFi.h>
#endif
#include <WiFiUdp.h>
#include <ArtnetWifi.h>
#include <Adafruit_NeoPixel.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#ifdef ESP32
  #include <SPIFFS.h>
#endif
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

//define your default values here, if there are different values in config.json, they are overwritten.
char numero_leds[3]= "50";
char numero_universo[2]= "2";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

// Neopixel settings
int numLeds = 2; // change for your setup
int numberOfChannels = numLeds * 3; // Total number of channels you want to receive (1 led = 3 channels)
const byte dataPin = 4;
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
Adafruit_NeoPixel leds = Adafruit_NeoPixel(numLeds, dataPin, NEO_GRB + NEO_KHZ400);

// Artnet settings
ArtnetWifi artnet;
int startUniverse = 2; // CHANGE FOR YOUR SETUP most software this is 1, some software send out artnet first universe as 0.

// Check if we got all universes
int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512) ? 1 : 0);
bool* universesReceived;
bool sendFrame = 1;
int previousDataLength = 0;

void fs_wifi ()
{
   //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");
[... 211 lines cut ...]
  numLeds = n_leds;
  numberOfChannels = numLeds * 3; 
  
  
  Serial.print("numLeds: ");
  Serial.println(numLeds);
  Serial.print("dataPin: ");
  Serial.println(dataPin);

  leds.updateLength(numLeds);
 
  artnet.begin();
  
  leds.begin();
  initTest();
  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);

}

void loop() {
  // put your main code here, to run repeatedly:
  artnet.read();
}
```

closed 2022-12-24 by rstephan