rstephan/ArtnetWifi #51: Unable to make a Artnet to DMX with esp_dmx

ElbyB, 2023-05-18:
Hi,

Maybe not really an issue, but is someone here can help me with this simple project?  I'm trying to use the esp_dmx library (https://github.com/someweisguy/esp_dmx) with the ArtnetWifi library to make a simple Art-Net to DMX node. I can easily send DMX with esp_dmx library and I can easily receive Art-Net with ArtnetWifi, but I'm unable to make them work together. Can someone here help me with this? Here is my code :

`/*

  DMX Write

  This sketch allows you to write DMX data to a DMX listener using a standard
  DMX shield, such SparkFun ESP32 Thing Plus DMX to LED Shield. This sketch was
  made for the Arduino framework!

  Created 10 September 2021
  By Mitch Weisbrod

  https://github.com/someweisguy/esp_dmx

*/
#include <Arduino.h>
#include <esp_dmx.h>

#include <ArtnetWifi.h>
#include <Arduino.h>

//Wifi settings
const char* ssid = "LRB_UBNT";
const char* pwd = "UBNT2LRB";
const IPAddress ip(192, 168, 1, 201);
const IPAddress gateway(192, 168, 1, 1);
const IPAddress subnet(255, 255, 255, 0);

WiFiUDP UdpSend;
ArtnetWifi artnet;
const int startUniverse = 0; // CHANGE FOR YOUR SETUP most software this is 1, some software send out artnet first universe as 0.

// Check if we got all universes
const int numberOfChannels = 512;
const int maxUniverses = numberOfChannels / 512 + ((numberOfChannels % 512) ? 1 : 0);
bool universesReceived[maxUniverses];
bool sendFrame = 1;

/* First, lets define the hardware pins that we are using with our ESP32. We
  need to define which pin is transmitting data and which pin is receiving data.
  DMX circuits also often need to be told when we are transmitting and when we
  are receiving data. We can do this by defining an enable pin. */
int transmitPin = 17;
int receivePin = 16;
int enablePin = 5;
/* Make sure to double-check that these pins are compatible with your ESP32!
  Some ESP32s, such as the ESP32-WROVER series, do not allow you to read or
  write data on pins 16 or 17, so it's always good to read the manuals. */

/* Next, lets decide which DMX port to use. The ESP32 has either 2 or 3 ports.
  Port 0 is typically used to transmit serial data back to your Serial Monitor,
  so we shouldn't use that port. Lets use port 1! */
dmx_port_t dmxPort = 1;

/* Now we want somewhere to store our DMX data. Since a single packet of DMX
  data can be up to 513 bytes long, we want our array to be at least that long.
  This library knows that the max DMX packet size is 513, so we can fill in the
  array size with `DMX_PACKET_SIZE`. */
byte dataDMX[DMX_PACKET_SIZE];

/* This variable will allow us to update our packet and print to the Serial
  Monitor at a regular interval. */
unsigned long lastUpdate = millis();

bool ConnectWifi(void)
{
  bool state = true;
  int i = 0;

  WiFi.begin(ssid, pwd);
  WiFi.config(ip, gateway, subnet);
  Serial.println("");
  Serial.println("Connecting to WiFi");
  
  // Wait for connection
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 20){
      state = false;
      break;
    }
    i++;
  }
  if (state) {
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  } else {
    Serial.println("");
    Serial.println("Connection failed.");
  }
  
  return state;
}

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  sendFrame = 1;
  
  // range check
  if (universe < startUniverse)
  {
    return;
  }
  uint8_t index = universe - startUniverse;
  if (index >= maxUniverses)
  {
    return;
  }

  // Store which universe has got in
  universesReceived[index] = true;

  for (int i = 0 ; i < maxUniverses ; i++)
  {
    if (!universesReceived[i])
    {
      sendFrame = 0;
      break;
    }
  }

  // read universe and put into the right part of the display buffer
  for (int i = 0; i < length; i++) {
      dataDMX[i] = data[i];
      
    }
    dmx_write(dmxPort, dataDMX, DMX_PACKET_SIZE);

  if (sendFrame)
  {
    
    dmx_send(dmxPort, DMX_PACKET_SIZE);
    dmx_wait_sent(dmxPort, DMX_TIMEOUT_TICK);

    Serial.print(" DataDMX (");

     for (int i = 0; i < 512; i++) {
      Serial.print(dataDMX[i], HEX);
      Serial.print(" ");
    }
    Serial.println();  

    // Reset universeReceived to 0

    memset(universesReceived, 0, maxUniverses);
  }

  
}

void setup() {
  /* Start the serial connection back to the computer so that we can log
    messages to the Serial Monitor. Lets set the baud rate to 115200. */
  Serial.begin(115200);
  ConnectWifi();
  artnet.begin();

  /* Set the DMX hardware pins to the pins that we want to use. */
  dmx_set_pin(dmxPort, transmitPin, receivePin, enablePin);

  /* Now we can install the DMX driver! We'll tell it which DMX port to use and
    which interrupt priority it should have. If you aren't sure which interrupt
    priority to use, you can use the macro `DMX_DEFAULT_INTR_FLAG` to set the
    interrupt to its default settings.*/
  dmx_driver_install(dmxPort, DMX_DEFAULT_INTR_FLAGS);

  // this will be called for each packet received
  artnet.setArtDmxCallback(onDmxFrame);
  
}

void loop() {
 
  artnet.read();
    
}`

Thnaks,

ElbyB, 2023-05-18:
Hi,

Found the problem, dataDMX[i] had to start at index [1] while data[i] had to start at index [0]...

for (int i = 0; i < length; i++) {
dataDMX[i+1] = data[i];

}

tonyxforce, 2023-10-01:
Can you please post the full working code?

ElbyB, 2023-10-01:
You have it in the fisrt post! Just make the change I mention in the second one...

tonyxforce, 2023-10-01:
Oh okay, thanks