rstephan/ArtnetWifi #53: BW16 (RTL8720DN)

Olivierdemoustier, 2023-07-22:
Hope this is the correct place to ask this. But could someone add the possibility to use this library on a BW16 . I really need 5Ghz. 2.4Ghz has to much interferences these days in TV studio's.
Thanks in advance. Olivier

rstephan, 2023-07-25:
Hi Olivier,

if it is a generic network/WiFi problem, than I am not the right one to blame.
Is the board supported by the Arduino library?
What support package do you use?
This one https://github.com/ambiot/ambd_arduino ?
Have you ask there for help?

I don't have a "BW16" board on hand yet.

My library uses the network stack (software stack) the board offers.
If there is no 5 GHz software support, you are out of luck.
Sorry.

Olivierdemoustier, 2023-07-27:
Hi,

 

I really don’t want to blame you for anything, I’, sorry if I have hurt your feelings.

It was just a question. My programming skills are limited. I have it working now, but not as a library. Just a standard UDP receiver and reading the whole package and then checking if it is artnet.....

But maybe this way, it will get slow if lots of traffic on the network?

 

Met vriendelijke groeten,

Best regards,

     

Olivier Demoustier

+32 475 451 257

 

This email has been checked for viruses by Avast antivirus software.
www.avast.com