rstephan/ArtnetWifi #58: Artnet Wifi connection lagging

Shotario, 2024-03-01:
Hello, I'm using esp32 wroom to light WS2812 strips from Resolume. Dev board has blue led, which indicates internet connection. When it's steady, lights are corresponding to signal. As soon as it's start blinking, led goes wrong, lagging. Even very close to router, notebook plugged to ethernet. I'm using single universe.  Connection isn't steady. Any suggestions?

rstephan, 2024-03-14:
How much LEDs are connected? Do you have a big enough power supply. The required current can add up quickly with 100 or more LEDs.

Shotario, 2024-03-14:
I was testing with a small amount, around 30 leds per strip. Ended up using
wled artnet and works flawless. I have no idea what caused signal lagging
and connection issues.

closed 2024-11-12 by rstephan