rstephan/ArtnetWifi #59: Send Artnet to many host

Remix4449, 2024-04-20:
j'ai actuellement plusieurs esp connecter entre elle par une esp en mode AP j'ai une esp qui transmet un trame artnet en broadcast. le problème c'est que les informations envoyées sacade.

lorsque je passe avec un diffusion dirigé vers une seul esp cela fonctionne 

j'aimerais envoyer une trame artnet à plusieurs host avec des adresses ip différents est ce que c'est possible ? 

Thanks you for your help :)

rstephan, 2024-04-22:
Hi,

you can try something like this:

```
String hosts[3] = {"192.168.0.1", "192.168.0.2", "192.168.0.7"};
...
void loop()
{
  for (i = 0; i < 3; i++) {
    artnet.begin(hosts[i]);
    ...
    artnet.setByte(1, 2); // or what ever color you like
    ...
    artnet.write();
    artnet.stop();
  }
}
```

Remix4449, 2024-04-30:
Good morning,

Thank you for your reply :)

I tried your solution, it works but unfortunately not for my project, we use switches and the program does all the universes one after the other by cutting the DMX frame at each receiver :(