rstephan/ArtnetWifi #60: error with #include <functional>

ChrisCrossTheUniverse, 2024-05-01:
Hi,

i'm getting this error. am i missing a library? what's it called

thanks

```
In file included from /private/var/folders/6g/9_f4whj51bz86jcc9nkwyjkh0000gn/T/.arduinoIDE-unsaved202441-8806-1wtc6o0.0pmdg/ArtnetWifiDebug/ArtnetWifiDebug.ino:7:0: /Users/.../Documents/Arduino/libraries/ArtnetWifi/src/ArtnetWifi.h:31:10: fatal error: functional: No such file or directory  #include <functional>           ^~~~~~~~~~~~ compilation terminated. exit status 1  Compilation error: exit status 1
```

ChrisCrossTheUniverse, 2024-05-16:
hi rstephan,

thanks for the reply. i use the arduino uno wifi rev2. i guess, it's not supported by this library. it works well with the nano esp32.

rstephan, 2024-05-17:
Hi,

you are right, the board is/was not supported. But long story short, I have included the megaavr line-up now!
It will take a day before the Arduino Library Manager will pick up the new version 1.6.0. But you can also use the code from github directly.

But by the way. You should have seen warnings like 'INCOMPATIBLE' and 'architecture(s) and may be incompatible with your current board'.

I don't have that board, so my testing was very limited. If it is working for you, give me feedback please.
Thanks.