rstephan/ArtnetWifi #61: How to get sender IP

Kabak85, 2024-10-10:
Hello,

I have a code using "  artnet.setArtDmxCallback(onDmxFrame);" function.

I need to get the sender IP for each frame/packet in the "onDmxFrame" function (filtering and merging purpose).

Is there a way to do or add this ?

Thanks.

rstephan, 2024-10-21:
Hi @Kabak85,

right now it is not possible to get that IP address information.
But it should be easy to that.
Give me some time and I will add that.

Kabak85, 2024-10-21:
@rstephan

Thanks, it will be very helpful.

rstephan, 2024-10-22:
Hi @Kabak85,

I was wrong, it is possible!
(The functionality was added 3 years ago and I must have overlooked it yesterday.)
You can request the value form the ArtNet-Object. See example below.

```
void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  IPAddress ip = artnet.getSenderIp();
  ...
}

```

Kabak85, 2024-10-23:
Hi @rstephan 

I tried with : 
`  Serial.print(IPAddress(artnet.getSenderIp()));
`
And it's working as expected. 

Thank for the help.