rstephan/ArtnetWifi #62: ArtNet Node or Device not discoverable

knopserl, 2024-10-22:
I'm trying to build an ESP32 (ESP32-S3) based ArtNet DMX node using this ArtnetWifi library.
I have tried several examples with the library but the device is not detected by my Artnet monitoring tools such as ArtNetominator.

I'm using the latest Arduino Studio version, an ESP32-S3 Board and the board is connecting to WiFi (ping also works).
I have one other AVR based ArtNet devices/node, which immediately shows up in ArtNetominator monitoring tool.

What am I doing wrong? Is this feature even implemented to publish/respond to an Artnet discovery/polling?
This is what Artnet devices usually should do.

Is there any sample code which should work and make the ESp32 discoverable/show up in Artnet clients app/tools under Windows 10? 

This is an example of a device which shows up in the monitor when it's powered up or if a query is sent:
[image]

rstephan, 2024-10-23:
Hi,
this is "by design". This library is as simple as possible.
If you need a node, take a look at my other project: https://github.com/rstephan/ArtnetnodeWifi

rstephan, 2024-10-30:
You can also use my library in "Ethernet-mode".
For a starting point take a look at this issue: https://github.com/rstephan/ArtnetWifi/issues/21
And if you don't like mine, use another one, no problem.